beagleboard/beaglebone-blue #7: Connection Refused 443

OscarJGV26, 2017-09-01:
Hi, 

I just bought a beaglebone blue, plugged it in and share my internet connection from the computer using /sbin/route add default gw 192.168.7.1 and changing the adapter settings to automatic ip configuration (beagle) and share in the ethernet adapter.

After that I was able to ping 8.8.8.8 and www.google.com. However, when I attempted to apt-get update install it gave me an error saying "Failed to connect to deb.nodesource.com port 443: Connection refused". 

I continued testing it and when I attempted to git clone some library, I was given the same "Failed to connect" on that same port. 

I shut down the windows firewall and my antivirus and even made some rules for the firewall to allow connections in/out to port 443 and nothing. 

I tried to check whether the system was listening on port 443 and this is what I got,

root@beaglebone:/var/lib/cloud9# netstat -plnt
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:53              0.0.0.0:*               LISTEN      5700/dnsmasq    
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      4436/sshd       
tcp6       0      0 :::8080                 :::*                    LISTEN      1323/apache2    
tcp6       0      0 :::80                   :::*                    LISTEN      1/init          
tcp6       0      0 :::53                   :::*                    LISTEN      5700/dnsmasq    
tcp6       0      0 :::22                   :::*                    LISTEN      4436/sshd       
tcp6       0      0 :::3000                 :::*                    LISTEN      1/init          
tcp6       0      0 :::1880                 :::*                    LISTEN      1/init 

Pleasee help. I don't know how to solve this and I am sure I am not the only one.

Regards,
Oscar

jadonk, 2017-09-20:
Windows can be a tricky beast. Can you repost to https://beagleboard.org/discuss? I feel like I've encountered this issue in the past and just had to look for more firewall-type settings.

OscarJGV26, 2017-09-20:
Thank you @jadonk 

I've already posted it (Connect refused on Port 443: Unable to apt-get update or git clone.) and found a temporary solution to the problem. Not sure if the Sharing "Settings" did the trick.

jadonk, 2017-12-19:
Personally, I find using 'sudo dhclient usb0' to work better than adding a route manually. This will fetch an IP address from your Windows machine, create the route and pull down the name servers. In other words, it will trigger a DHCP request.

jadonk, 2017-12-19:
If you need further help, please provide the link to where you posted it.

closed 2017-12-19 by jadonk