beagleboard/beaglebone-blue #9: Suggestion: mention motor driver specs in spec sheet

ElhemEnohpi, 2017-09-15:
It would be convenient if the "BeagleBone_Blue_ShortSpec.pdf" and the website could mention the specs of the motor drivers, i.e., 1.2A continuous, 3.2A peak, 2.5-13.5V; and/or the part number TB6612FNG. Thank-you.

closed 2017-09-20 by jadonk

ElhemEnohpi, 2017-09-20:
I don't get what you did there. The .docx file you added just says the same as the pdf, "4 DC motor drivers". It doesn't mention the power (current & voltage) specs.

AJRepo, 2017-10-20:
@ElhemEnohpi  

This was super helpful, thanks, LMNOP! I found a PDF for the specs for the [TB6612FNG](http://www.pololu.com/file/0J86/TB6612FNG.pdf) H-Bridge which lists the following 

- Power supply voltage; VM = 15  V(Max)
- Motor supply (2.5 V~13.5 V)
- Output current; IOUT=1.2 A(ave) / 3.2 A (peak) per channel
- Output low ON resistor; 0.5Ω (upper+lower Typ. @VM ≥ 5 V)

Also suggest adding to the spec (for noobs like me) noting that there are **two** of them on the board, each driving two of the DC motor drivers (M1,M2) and (M3,M4).

ElhemEnohpi, 2017-11-02:
@jadonk I'm a bit confused, your commit of BeagleBone_Blue_ShortSpec.docx says "Add motor driver strength", but the document still doesn't mention the motor driver strength. Maybe you put up the wrong version?

filmo, 2017-11-04:
I would add Output Currents mentioned by AJRepo are 'per-channel' there are four channels on the board, two per chip.