beagleboard/beaglebone-blue #12: I am not sure wheather I am allowed to ask this question here, but any help is pretty much aprechiated.

BluePackStack, 2017-11-05:


RobertCNelson, 2017-11-06:
@BluePackStack all the out of box control of wl18xx for the default image is located in:

```
/etc/default/bb-wl18xx
```
Regards,

BluePackStack, 2017-11-07:
After restarting. Changes in `/etc/default/bb-wl18xx` are finally done. Very nice!

 Is it true that connman is very buggy on beaglebone? Is there an alternative you can recommend?

RobertCNelson, 2017-11-07:
It's not buggy, just highly intergrated for the out of box experience... I've setup /etc/default/bb-wl18xx so you can disable parts of it..

BluePackStack, 2017-11-07:
> It's not buggy, just highly intergrated for the out of box experience... I've setup /etc/default/bb-wl18xx so you can disable parts of it..<

I cleared my early response and went to bed then, since I recognized, that the board wouldn't response to ssh anymore after editing the settings. It seems I can sometimes manipulate softAp SSID and password through different ways, but at the cost of losing control over the board every time. Always have to rewrite SD/Memory to get into system again.

Using: bone-debian-9.2-iot-armhf-2017-10-10-4

Anyway I am very happy that you help me! I will add further information about what exactly crashes my board.

closed 2017-11-07 by BluePackStack