beagleboard/beaglebone-blue #21: Losing Connection with Satellite DSMX recv

prashinz, 2019-04-14:
I am using  DSMX satellite recv on my beaglebone blue and have a DX6 transmitter, which is binded to sat recv via DSM2 protocol running (rc_bind_dsm). After few minutes of binding , I  loose DSM connection and the satellite recv starts blinking. I have no clue whats triggering it.  

Any suggestions / help would be appreciated. 

Thanks.

jadonk, 2019-06-03:
Not sure about what blinking means on this recv'r. Is there an environment where you can get it to work?

prashinz, 2019-06-05:
So I have noticed that whenever I connect my BBBL to internet, I frequently loose connection with DX6 transmitter. As soon as I disconnect from internet the connection I never loose connection with DX6.