beagleboard/beaglebone-blue #24: Blue will not generate a WiFi AP on boot

dmalawey, 2019-05-19:
On many occasions my students powered up their device and failed to see a wifi signal.  NEITHER THE PROBLEM NOR MY SOLUTIONS CAN BE CONSISTENTLY REPEATED, but it has recurred over several months with 10-20 different units with different images flashed, so I need help.  I gave the following instructions to make sure wifi is running:

1) Be sure to power with a 12v source at barrell plug instead of USB, which may be insufficient to run wifi at times. (is this true???) - this sometimes resolved the problem.
2) Shut down your blue, remove power, remove SD card and replace SD card.  Boot back up and check for wifi functionality. - this sometimes resolved the problem.

jadonk, 2019-05-30:
> On many occasions my students powered up their device and failed to see a wifi signal. 

What do you mean by "failed to see a wifi signal"? How are you checking? Do you mean they don't see the access point SSID?

> NEITHER THE PROBLEM NOR MY SOLUTIONS CAN BE CONSISTENTLY REPEATED, but it has recurred over several months with 10-20 different units with different images flashed, so I need help. I gave the [...]

USB should be sufficient for powering WiFi. Powering off 12V (really 9V-18V) without a battery installed isn't fully supported. Unlikely this is the problem.

> 2. Shut down your blue, remove power, remove SD card and replace SD card.  Boot back up and check for wifi functionality. - this sometimes resolved the problem.

So, you are running off of microSD? I suspect the issue is likely related to the bootloader on the eMMC not matching the image on the microSD card.

dmalawey, 2019-11-13:
Reboot sequences have been more reliable.  

SOLUTION: One or two of our Beagles (from the class) were judged to have hardware corruption and were taken out of circulation.  Probably this was due to student misuse but the damage is not visible.  Transferring a known-working SD image from one good board to the problem board is our method to verify if hardware has a problem.    It's not reliable to try to get the problem to express itself, reliably, otherwise.

closed 2019-11-13 by dmalawey