beagleboard/beaglebone-blue #26: Identification of motor driver pins

dmalawey, 2019-06-06:
I have not found a document that tells us which pin on the motor driver schematic corresponds to pins 1 or 0 of each motor driver.   Where can we find this?

[image]

For example.  Is motor 1 pin 1 located on the lower or upper pin in this photo?

[image]

jadonk, 2019-06-11:
You said "pins 1 or 0", but the connectors label the pins 1 and 2. It is common for hardware to begin indexes at 1 instead of 0.

The hardware design convention is pin 1 gets the square solder pad.  You can find the solder pad from the bottom of the board or looking at [the board layout](https://github.com/beagleboard/beaglebone-blue/blob/master/BeagleBone_Blue_brd.pdf).

[image]

The silkscreen attempts to communicate the clockwise numbering of the motor connectors, starting at the lower-right in your picture.

So, motor 1 pin 1 is the lower-right connector on the upper-most pin within that connector, whereas motor 1 pin 2 is the lower-right connector on the lower-most pin within that connector.

There is a nice TB6612FNG motor driver truth-table on [Sparkfun's website](https://learn.sparkfun.com/tutorials/tb6612fng-hookup-guide/all). I've mapped the pins to the gpio/pwm numbers as defined in the [librobotcontrol motor driver](https://github.com/StrawsonDesign/librobotcontrol/blob/master/library/src/motor.c) and schematic processor ball numbers below:

| Mx | In1 | In2 | PWM | Out1 | Out2 | SW correction |
| -- | -- | -- | -- | -- | -- | -- |
| M1 | GPIO2.0 T13 | GPIO0.31 U17 | EHRPWM1A U14 | 2 | 1 | -1 |
| M2 | GPIO1.16 R13 | GPIO0.10 U17 | EHRPWM1B T14 | 1 | 2 | 1 |
| M3 | GPIO2.9 R4 | GPIO2.8 R3 | EHRPWM2A U10 | 1 | 2 | 1 |
| M4 | GPIO2.6 R1 | GPIO2.7 R2 | EHRPWM2B T10 | 2 | 1 | -1 |

Note that the software accounts for when Out1 is connected to pin 2 by performing a correction.

dmalawey, 2019-06-11:
This helps very much!  Will add this to our robot slides. Thank you.

closed 2019-06-11 by dmalawey