beagleboard/beaglebone-blue #33: The two black JST-SH connectors don't have a retention bed, unlike the white JST-SH connectors

dorovl, 2019-12-09:
The connector that plugs inside is not retained. By vibrations, this connector comes out and this is particularly dangerous in the context of a flight controller (our project). 

- **Therefore we recommend to put also a white JST-SH connectors on the next version.**

dmalawey, 2019-12-09:
I have had a problem with the white plastic housings on the motor drivers pulling right off the board when we pull a plug out.   They are not JST-SH, I can't recall the spec.

dorovl, 2019-12-09:
Please note I'm speaking about the "Power out" and "4 ADC" connectors (see red arrow) :

[image]

I also had the problem with the plastic housings of the motor drivers pulling right off the board when I pulled out a plug, but maybe you might open another issue for this?

dorovl, 2019-12-10:
Closed, because it was submitted on upstream git repository (jadonk/beaglebone-blue) instead.

jadonk, 2020-06-09:
Let's keep the issues here.