beagleboard/beaglebone-blue #39: Do we keep WL1835 or move to WL1831?

jadonk, 2020-06-09:
---

Migrated from https://github.com/jadonk/beaglebone-blue-jadonk/issues/30
Originally created by @jadonk on *Wed, 12 Oct 2016 21:08:39 GMT*

---

jadonk, 2020-06-17:
For rev C, I'd suggest moving to WL1807, dropping Bluetooth and adding 5GHz. Unsure on availability.

dorovl, 2020-06-17:
> For rev C, I'd suggest moving to WL1807, dropping Bluetooth and adding 5GHz. Unsure on availability.

Hi @jadonk I'm using Bluetooth to communicate with deceives like Arduinos having a simple Bluetooth modem (for telemetry purposes). For me, dropping Bluetooth would be a little regrettable because I should to add more wires also level sifters (3.3V ↔ 5V) for the UARTs.