beagleboard/beaglebone-blue #43: MPU 9250 via SPI

jadonk, 2020-06-09:
---

Migrated from https://github.com/jadonk/beaglebone-blue-jadonk/issues/41
Originally created by @mirkix on *Fri, 23 Dec 2016 22:10:42 GMT*

---
Hi, is it an option to connect the MPU 9250 via SPI and not via I2C? With Ardupilot we can sample at 4 kHz / 8 kHz (acc gyro) in this case I2C is not fast enough for the data transfer. It is possible to sample slower too, than the transfer speed via I2C should be okay.

jadonk, 2020-06-09:
---

Migrated from https://github.com/jadonk/beaglebone-blue-jadonk/issues/41#issuecomment-269523192
Originally created by @StrawsonDesign on *Wed, 28 Dec 2016 18:59:33 GMT*

---
Hi Mirko,

Sadly the 9250 is on the I2C bus on both Blue and Cape. In my flight code I
use the 9250's DMP for quaternion orientation estimation and also use the
DMP's fixed-rate interrupt function (up to 200hz only) to generate the
timing for the discrete feedback loop. Pulling accel, gyro, mag, and
quaternion data at 200hz puts the i2c bus at about %15 use when taking
advantage of the 9250's FIFO buffer to do one big read at a time.

I have no doubt sampling just the accel, and gyro 1khz would be possible
over I2C with intermittent reads to the magnetometer. However, even on a
small quadcopter the crossover frequency for the roll/pitch feedback
controllers is only around 10-20hz so I've found 200hz sample rate to be
more than sufficient for smooth feedback control. Anything faster is
difficult to achieve in userspace linux without multiple cores anyway.

Best,
James

Dept. Mechanical & Aerospace Engineering
University of California, San Diego

jadonk, 2020-06-09:
---

Migrated from https://github.com/jadonk/beaglebone-blue-jadonk/issues/41#issuecomment-272521425
Originally created by @jadonk on *Fri, 13 Jan 2017 19:11:09 GMT*

---
@mirkix @StrawsonDesign shall I try to ping Invensense on the DMP documentation issue?

jadonk, 2020-06-09:
---

Migrated from https://github.com/jadonk/beaglebone-blue-jadonk/issues/41#issuecomment-269560461
Originally created by @mirkix on *Wed, 28 Dec 2016 23:52:59 GMT*

---
@StrawsonDesign Hi James, is there any documentation free available about the DMP? I think ArduPilot try to sample as fast as possible to avoid problems caused by vibrations (aliasing).
I am working on ArduPilot support for the Blue and made good progress so far.

Regards
Mirko

jadonk, 2020-06-09:
---

Migrated from https://github.com/jadonk/beaglebone-blue-jadonk/issues/41#issuecomment-272566211
Originally created by @StrawsonDesign on *Fri, 13 Jan 2017 22:32:30 GMT*

---
Hi Mirko, Jason,

Free documentation on the DMP is limited to their example code and a
pamphlet with a brief overview of functionality. I've played with the DMP a
lot over the past few years and I think I've gotten as much performance out
of it as I can having written my implementation %75 from scratch. I even
managed to get it to package the magnetometer data into the same FIFO
packet as the DMP, accel, and gyro data which a comment in Pansenti's
mpu9250 driver said was "impossible". The sparkfun library is horrifically
slow and does a lot of unnecessary register modifications every time it
swaps between reading mag and dmp.

The DMP is internally limited to 200hz, presumably because it has to do a
LOT of floating point arithmetic for it's quaternion estimate. It always
samples and runs its filter at 200hz, even if you set up the interrupt
signal to run as a divisor of that. Sadly, that also means that the accel
and gyros have to be set to sample at 200hz too, so pinging them between
DMP reads is useless.

I understand your worry about aliasing, but 200hz is still 20x faster than
the crossover frequency of even my fastest feedback controller. A properly
designed compensator (eg not PID!) will have gain roll-off well before
200hz.

Best,
James

jadonk, 2020-06-09:
---

Migrated from https://github.com/jadonk/beaglebone-blue-jadonk/issues/41#issuecomment-275246828
Originally created by @mirkix on *Wed, 25 Jan 2017 22:02:22 GMT*

---
@jadonk :+1:

jadonk, 2020-06-09:
---

Migrated from https://github.com/jadonk/beaglebone-blue-jadonk/issues/41#issuecomment-274890112
Originally created by @jadonk on *Tue, 24 Jan 2017 18:21:23 GMT*

---
@mirkix Can you confirm you are comfortable with @StrawsonDesign's answer?

dorovl, 2020-06-17:
@mirkix: I don't saw any answer from you confirming if the MPU 9250 via I2C is fast enough or not, i.e. can you confirm you are comfortable with @StrawsonDesign's answer? As you know, the Arduplot is based on a PID controller. My flight controller has just been installed and setup on my nitro helicopter; I will make a flight test soon and will be able to provide some data to analyze.

@jadonk: I'm not a specialist of the MPU9250, but I saw that on TDK's site, the MPU9250 is not recommended for new designs: https://invensense.tdk.com/products/motion-tracking/9-axis/mpu-9250/. If this device is superseded, maybe it's possible to integrate the new device on this revision? Otherwise, it would ideal be to have multiple 9DOF IMUs (by instance MPU9250 and LSM9DS1)