beagleboard/beaglebone-blue #51: Add PDFs

jadonk, 2020-06-09:
---

Migrated from https://github.com/jadonk/beaglebone-blue-jadonk/issues/9
Originally created by @jadonk on *Thu, 03 Dec 2015 22:16:52 GMT*

---

jadonk, 2020-06-09:
---

Migrated from https://github.com/jadonk/beaglebone-blue-jadonk/issues/9#issuecomment-378816201
Originally created by @silver2row on *Thu, 05 Apr 2018 04:20:49 GMT*

---
Hello...I see you want help. It is late now but I will have time every now and then. Better late than never.

Seth

P.S. Reply if you still can use someone to assist.

jadonk, 2020-06-09:
---

Migrated from https://github.com/jadonk/beaglebone-blue-jadonk/issues/9#issuecomment-248329459
Originally created by @jadonk on *Tue, 20 Sep 2016 15:04:44 GMT*

---
Need to automate the generation to stay up-to-date. The OSHPark preview looks really great.

silver2row, 2020-07-24:
Hello,

Would anyone like to see items fixed to .pdf format? I have some .pdf maker on Win/Linux. 

Seth

dmalawey, 2020-07-25:
> Hello,
> Would anyone like to see items fixed to .pdf format? I have some .pdf maker on Win/Linux.
> Seth

We waited all our lives for PDF instructions.  -Texas A&M Beaglebone users