beagleboard/beaglebone-blue #58: [Help Wanted] BeagleBone Blue Does Not Turn On (No LEDs nor USB Detection)

TiberiusDreadnought, 2022-07-17:
### Expected Behavior

Plugging in USB or 12V barrel jack power supply to BeagleBone Blue (BBBL) should

1. Enable the user LEDs;
2. Turn the "ON" LED on;
3. Be recognizable by Windows (if plugged in by USB).

### Actual Behavior

Using the USB or 12V barrel jack power supply has the following characteristics:

- *Momentarily flashes once* the "ON" LED;
- All other LEDs are off for the entire duration;
- When *specifically* the USB is plugged in:
  - No USB is detected by Windows.;
  - SSH at 192.168.7.2 fails;
  - Pinging 192.168.7.2 fails.
- When *specifically* the 12V barrel jack is plugged in:
  - The "CHG" LED glows for the entire duration.

This issue began when the 12V/2A barrel jack power supply was plugged into a powered off BBBL. At that time, the BBBL LEDs glowed as usual, and immediately afterwards, turned off (excluding the "CHG" LED). 

**Note:** The barrel jack power supply was previously used with no concerns.

### Attempted Solutions

- Holding down the SD (boot select) button while powering on (either by inserting USB, barrel jack, or by pressing the POW (power) button) when a bootable microSD card is present.
- Holding down the POW button.
- Holding down both SD and POW button at the same time or one after another and other button pressing configurations.
- Pressing the RST button.
- Reflashing the SD card with the latest image (note that the eMMC already had the latest image) and then inserting it into to the BBBL

Using a multimeter, 5V was detected across the GPS port (pins identified from the PINOUT section of this wiki) suggesting that the whole circuitry may not be dead.

### Specifications

| Location | Image name |
| - | - |
| On-board image | bone-debian-10.3-iot-armhf-2020-04-06-4gb.img |
| microSD image (before re-flash)* | bone-ubuntu-18.04.3-ros-iot-armhf-2019-12-16-6gb-LLW-2019-12-23.img |
| microSD image (after re-flash) | bone-debian-10.3-iot-armhf-2020-04-06-4gb.img |

*\* Originally was used when the problem first occurred; This image was previously used with no concerns.*

**Note:** This issue maybe a duplicate of [this older post](https://github.com/beagleboard/beaglebone-blue/issues/19#issue-424323908) which was unanswered.

aritomas, 2022-11-23:
I have the same problem. Have you solved it?

TiberiusDreadnought, 2022-11-24:
@aritomas , unfortunately not.